scorelab/DroneSym #55: Dockers should have proper Label Schema labels

tanmayanand44, 2017-12-11:
This repos has a Dockerfile but it does not contain any Label Schema labels.  Following are the lables we have to have on our Dockers. 
- [ ] `name` 
- [ ] `description` Shoud be on user's perspective.
- [ ] `url` Should be SCoRe website
- [ ] `vcs-url` Should point to GitHub repo
- [ ] `vcs-ref`
- [ ] `vendor` Shoudl be "SCoRe Lab"
- [ ] `version`
- [ ] `schema-version`

agentmilindu, 2017-12-11:
Dockers can be found in `docker` branch (  https://github.com/scorelab/DroneSym/tree/docker )

Kalyan3578, 2017-12-26:
I'm going to work on one of the dockerfile!

agentmilindu, 2017-12-26:
Nice! Pick one and tell us.

Kalyan3578, 2017-12-26:
@agentmilindu I have already started working on this dockerfile: https://github.com/scorelab/DroneSym/blob/docker/dronesym-python/Dockerfile

agentmilindu, 2017-12-27:
@tanmayanand44 Nice!

closed 2017-12-31 by malithsen