scorelab/DroneSym #59: Package.json Issues

iammosespaulr, 2017-12-31:
{
  "valid": true,
  "warnings": [
    "Missing recommended field: description",
    "Missing recommended field: keywords",
    "Missing recommended field: bugs",
    "Missing recommended field: author",
    "Missing recommended field: contributors",
    "Missing recommended field: repository"
  ],
  "recommendations": [
    "Missing optional field: homepage",
    "Missing optional field: engines"
  ]
}

iammosespaulr, 2018-01-02:
MEANINGFUL DESCRIPTION :
to solve the warning one must add the description field, the keyword field, the bugs field, the author field, the contributor's field, the repository field and the homepage field.

malithsen, 2018-01-02:
@iammosespaulr Thanks. Feel free to send a PR