scorelab/DroneSym #63: New Features for DroneSym

iammosespaulr, 2018-01-02:
[image]
 
The Login Page Has Only The Words Login It could be Made into LOGIN TO DRONESYM or something like that

iammosespaulr, 2018-01-02:
We need to have an improved login Page as well as a better DashBoard as the current one is not that good  We Should Have the options to choose the MAP section.
[image]

malithsen, 2018-01-02:
>We Should Have the options to choose the MAP section rather than have it stay all the time.

Hi @iammosespaulr could you please elaborate this

iammosespaulr, 2018-01-02:
@malithsen  What I wanted to say was that the map remains open all the time ,we should have some kind of toggle option for the map

malithsen, 2018-01-02:
I think it's sensible for the main view to have the map opened by default because that's a what a user would use for the majority of the time with dronesym.

iammosespaulr, 2018-01-02:
Yeah It makes sense but i was thinking from the perspective of a person who would like to save mobile data and computing power

iammosespaulr, 2018-01-02:
Why not have the drone's camera connected???                <---- New feature

iammosespaulr, 2018-01-02:
Use multiple drone cameras To stitch a huge picture using a python script and have it broadcast on the screen .. This would give you live feed of a huge area

iammosespaulr, 2018-01-02:
Most drones these days are capable of broadcasting live video feeds and based on the drone's position we could write a script that would stitch the live feeds to help us monitor huge areas.

iammosespaulr, 2018-01-02:
@malithsen Would you approve my GCI task ?

malithsen, 2018-01-03:
1. While it could be useful for certain users to have stitched live feeds, I believe such a feature would be out of the scope of dronesym's intended use cases. Viewing real-time data individually on the other hand makes more sense #60 
2. Please open a new issue for new suggestions

closed 2018-01-03 by malithsen