scorelab/DroneSym #73: New Feature For dronesym

iammosespaulr, 2018-01-05:
try to add a feature that allows login with OAuth 
Login With Google
Login With Facebook
Login With Github 
and save all the drone data for a particular account such that it can be accessed by just logging into the
respective accounts.

iammosespaulr, 2018-01-05:
@malithsen  What do you think of this one ?
Pls do review my GCI task

malithsen, 2018-01-06:
Hi @iammosespaulr,
Good suggestion. Please change the title to convey the overall idea you're suggesting.

VibhorCodecianGupta, 2018-10-01:
#95 fixes this issue to an extent. The PR just includes the backend framework for now, will gladly work on the frontend once this gets approved.

What it does:
Allows login functionality to dronesym, and once the user is logged in, allows them to connect their other social accounts.

@agentmilindu please review :)

dilinade, 2019-02-17:
I'm working on this issue.

VibhorCodecianGupta, 2019-02-17:
@dilinade hey! I'm almost done with the frontend part, and the backend is already implemented. Could you give me a day or two? Running busy due to some college stuff, wouldn't want both of us to do the redundant tasks. You can take up some other issue perhaps, in the meanwhile? Thanks

dilinade, 2019-02-17:
@VibhorCodecianGupta Sure thing then. I will look on to another issue. Thanks for informing

dilinade, 2019-03-16:
@VibhorCodecianGupta Hi, Are you done with this issue?

Radesh-kumar, 2021-02-02:
@iammosespaulr @ivantha I'd like to work on this issue if it is still open
Thank you!