scorelab/DroneSym #104: i) The files in the folder should be arranged more properly

Deep-Chandra, 2018-10-27:
# Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I'm reporting the issue to the correct repository (for multi-repository projects)

# Expected Behavior

Please describe the behavior you are expecting

# Current Behavior

What is the current behavior?

# Failure Information (for bugs)

Please help provide information about the failure if this is a bug. If it is not a bug, please remove the rest of this template.

## Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. step 1
2. step 2
3. you get it...

## Context

Please provide any relevant information about your setup. This is important in case the issue is not reproducible except for under certain conditions.

## Failure Logs

Please include any relevant log snippets or files here.

Deep-Chandra, 2018-10-27:
While creating a drone the user must have the ability to name his/her drone. This is something that everyone wishes to have. Also, it would be better if the files were arranged more properly.

ivantha, 2019-02-13:
Duplicate issue : #105

closed 2019-02-13 by ivantha