scorelab/DroneSym #107: Improvements and issues

kunalbhatia18, 2018-10-27:
# Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
- [ ] I checked the documentation and found no answer
- [ ] I checked to make sure that this issue has not already been filed
- [ ] I'm reporting the issue to the correct repository (for multi-repository projects)

# Expected Behavior

It should follow the gitflow workflow that is:
=>Require status checks to pass before merging is active
=> Require branches to be up to date before mergin
=> Appropriate Status checks are required
=> Enforce all configured restrictions for administrators is active
 =>Restrict only maintainers group can push to the master branch

# Current Behavior

=>many syntax errors in code
=>the files in the folder are all unscrambled, it should be more systematic

# Failure Information (for bugs)

there's a bug on user groups

## Steps to Reproduce

Please provide detailed steps for reproducing the issue.

=>you could improve the GUI/UI/UX

malithsen, 2018-10-27:
Hi @kunalbhatia18
Please be specific on the errors you encountered. 
Why do you think the folder structure is unsystematic.

> there's a bug on user groups
What exactly is the bug? and how to reproduce it

> you could improve the GUI/UI/UX
What are the places you find that has bad UI?

Please open separate issues for each issue if you have many

kunalbhatia18, 2018-11-05:
Make it Python 3 Compatible , life of Python 2.7 is about to end

ivantha, 2019-02-13:
I am closing this issue because it does not seem relevant anymore.

closed 2019-02-13 by ivantha