scorelab/DroneSym #110: [GCI] Switch from google maps to open source alternative

tintinmovie, 2018-10-28:
# Prerequisites

- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
- [x] I'm reporting the issue to the correct repository (for multi-repository projects)

# Current Behavior
Current project is using the google maps api. Google maps api is not free, and will cost money if you go over the set allowance per api key. 

# Expected Behavior/ Idea
Using an opensource alternatives - such as leafletjs for the slippy map interface, and openstreetmaps for the tile server, you can migrate from google maps to a fully opensource alternative that is completely free (the openstreetmaps tile server doesn't even require an api key)

dilinade, 2019-08-16:
The best alternative that we can find is OpenStreetMaps (OSM) which still has a low data quality which is spotty. Therefore, in the current development phase it's better to go with Google Maps API for the time being from my perspective.

ivantha, 2019-09-03:
@charithccmc @rhperera @malithsen  Would we need this in the long run? 
I agree with @dilinade that OSM or any or the current open source alternatives has low quality data compared to Google Maps. It is hard to imagine that the situation would reverse.

tintinmovie, 2019-09-03:
Maybe integrate both map api's? Use OSM for development, and when DroneSym receives adequate funding switch to Google Maps for higher quality maps that users would prefer.

ivantha, 2019-09-11:
@tintinmovie That seems like a lot of redundant work. As it looks like there is not much interest on this particular topic of discussion, I am closing this issue until it becomes a real issue.

closed 2019-09-11 by ivantha