scorelab/DroneSym #111: change README for db.js, Provide Firebase instructions

anuragparida, 2018-10-28:
- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed
- [x] I'm reporting the issue to the correct repository (for multi-repository projects)

# Expected Behavior

* To find and edit `db.js` in `DroneSym/dronesym-node` folder.

# Current Behavior

* `db.js` is replaced in the repo by `example.db.js`. The `README` should be updated to highlight the change because new users may not be able to understand the change.
[image]

* Most users have not used `FirebaseAdminSDK` previously before so documentation has to be provided to help new users to setup their `firebase key file` and `database url`.
## Suggestion-
* The instructions at [FireBaseSetup](https://firebase.google.com/docs/admin/setup#add_firebase_to_your_app) have to explained in a basic manner in the `README`/project wiki.

OshanIvantha, 2018-11-24:
@anuragparida You are welcome to do a PR on the issues that you highlighted 🙂

anuragparida, 2018-11-24:
I'll get right to it! :smile: