scorelab/DroneSym #131: No Forgot Password option

hcktheheaven, 2018-11-11:
# Expected Behavior

There should be a button on Login Page which send password reset email to the respective user which redirects to password reset form.

# Current Behavior

There's no such feature available.

meetcric, 2019-02-17:
I would like to work on this issue..@ivantha

meetcric, 2019-02-19:
To implement "forget-password" feature I need an email-id from which we sent a new password to the user. 
so can you give me email-id or for now i will use mine?@ivantha

Anmolbansal1, 2019-03-03:
@meetcric are you still working on this??

meetcric, 2019-03-04:
Yes @Anmolbansal1

meetcric, 2019-03-04:
@ivantha I think it is a stale issue, should we close it or I make a PR for this?

ivantha, 2019-03-04:
@meetcric Did you implement the functionality?

meetcric, 2019-03-04:
@ivantha actually I am on the halfway.

ivantha, 2019-03-13:
@meetcric It would be better if you could complete it 🙂

PR #365 (2019-08-18): Forgot password done
Finished forgot password component.

Note : Password reset email is being sent using Ethereal email which is a temporary SMTP service, mostly aimed at Nodemailer. (Email link can be found in the node console)

Closes #131
closed 2019-08-18 by ivantha