scorelab/DroneSym #153: [UX] Active Route is not indicated

TheLukaszNs, 2018-11-28:
# Current Behavior
The Navbar and Sidebar (at the `dashboard/user` route) don't show which route is selected. It's not a big issue, but we can slightly improve User Experience by indicating on what page the user currently is.
[image]

# Expected Behavior
The Links in the Navbar and Sidebar should be showing in which route the user currently is.

## Steps to Reproduce
1. Login to *DroneSym*
2. Go to the *Dashboard* section (or any other)

meetcric, 2019-02-03:
@ivantha should I use Angular2 Breadcrumb for the user navigation on the website?

ivantha, 2019-02-05:
@meetcric You can use any method you want to indicate the route of the user. Maybe something like breadcrumbs etc.

PR #167 (2019-02-15): Feat:Add breadcrumb to indicate the route of user
This PR  improve User Experience by indicating on what page the user currently is.I use  Breadcurmbs to indicate the route of the user.

[image]

 `closes #153 `
closed 2019-02-15 by ivantha