scorelab/DroneSym #165: Can't use <center> tags in a front-end .html file

rohancl, 2018-12-14:
# Expected Behavior

<b>Proper and modified Output on using </b>```<center>```<b> tags in HTML </b>
<ul>
<li><b>Issue File Location: </b> dronesym-frontend/src/app/login/login.component.html</li>
</ul>

# Current Behavior

What is the current behavior?
Currently, the use of ```<center>``` tags in html reloads and Dronesym page and shows nothing

[image]

# Failure Information (for bugs)
Let's go through a flow and understand this issue in detail by understanding it step by step....<br>
After cloning the repository, installing all the dependencies and setting it up. I tried it's functioning and all the other things. It was working perfectly fine. 
<b>
1. I tried editing the login.component.html file to change it's for any frontend betterment possibilities.
2. In one case, I had to use the ```<center>``` tags in the code.
3. After placing the tags, The page reloading and nothing appeared.

Conclusion: I tried editing that .html file with other things such as ```<br>```, The page reloaded and it also did work.

[image]

This concludes that the droneSym front end .html files are updated when modified, except if we use ```<center>``` tags in the code..</b>

## Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. Install and Setup DroneSym locally.
2. Go into ```dronesym-frontend/src/app/login/login.component.html```
3. Try to use ```<center>``` tag anywhere in that HTML file.
4. Notice that the screen now gives no output 

<h2>NOTE:</h2> This is just oneof the html files which I've checked, I did try some more .html files, This issue with present with them too.. and might be present with others too...

harsha-iiiv, 2019-01-10:
Can I work on this issue?

dilinade, 2019-02-12:
<center> tags are deprecated. It's been removed from Web Standards though some browsers still support it. This may cease to work anytime.
Reference : 
https://developer.mozilla.org/en-US/docs/Web/HTML/Element/center

rohancl, 2019-02-14:
A regular HTML file, when opened in the firefox works well with the center tags. Is there something special about this file? @dilinade

dilinade, 2019-02-19:
@rohancl From HTML5 onwards the center tag is deprecated and it's being removed by most of the browsers. But some browsers do still support it. And I think that's the case.

[image]

ivantha, 2019-02-19:
@rohancl I am closing this issue as it seems the confusion has been cleared. 
Please reopen if there are further queries.

closed 2019-02-19 by ivantha