scorelab/DroneSym #251: Bug   Comapny logo redirects to login page

prijeshb, 2019-03-30:
- [x] I am running the latest version
- [x] I checked the documentation and found no answer
- [x] I checked to make sure that this issue has not already been filed

# Expected Behavior

On clicking DroneSym Logo name user must be redirected to homepage.

# Current Behavior

It is redirected to login page instead.

## Steps to Reproduce

Please provide detailed steps for reproducing the issue.

1. click on DroneSym after logging or signing up on dashborad

dilinade, 2019-03-31:
I would like to work on this issue

prijeshb, 2019-03-31:
@dilinade Alreay working on it !!

dilinade, 2019-04-01:
@prijeshb Are you done with this?
I just saw the comment and I'm done.

prijeshb, 2019-04-01:
I have already done it but forgot to make pull request ;(

PR #252 (2019-04-01): Bug on Logo
Fixes the click on DroneSym which redirected to the login page.

Closes #251
closed 2019-04-01 by ivantha