scorelab/DroneSym #440: Docker install error on windows

cjprogamer, 2020-01-10:
**Error when try to install Dronesym with docker**

When the command `docker-compose up` executed in the project root, It gives this error.

[image]

My environment - 
       

- OS - Windows 10 1903

- Docker version -  19.03.5, build 633a0ea

- Node - v13.5.0

ivantha, 2020-01-10:
@dilinade Do you know what causes this?

dilinade, 2020-01-10:
@cjprogamer Does _docker ps_ working fine? And is virtualization enabled in your machine?
@ivantha I think it's an issue due to the local environment

JosephSemrai, 2020-01-12:
@dilinade I believe it may be a config issue. The docker build is successful for me on Windows Build 18363:
[image]

dilinade, 2020-01-17:
@JosephSemrai So it builds successfully  right?

JosephSemrai, 2020-01-17:
@dilinade Yes, it builds for me on Windows.

closed 2020-01-17 by dilinade